jneilliii/OctoPrint-BedLevelVisualizer #623: [FR]:  A delay time while it preheats the bed to measure the mesh

erikcsellers, 2023-07-13:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

erikcsellers, 2023-07-13:
no problem, just would be nice if it was actually to temperature before it starts.

jneilliii, 2023-07-13:
not sure I understand what you mean?

erikcsellers, 2023-07-13:
Sorry never mind, I guess it already is preheating the bed. I was just impatient

closed 2023-07-13 by erikcsellers

jneilliii, 2023-07-13:
Yeah, preheating the bed is just another gcode line you put in the settings (prior to `@BEDLEVELVISUALIZER`), so as long as you put the correct commands it should do that for you. If you want to sit on a specific temperature after reaching temp you could probably use a dwell command [G4](https://reprap.org/wiki/G-code#G4:_Dwell) to basically tell the printer to wait a period of time before running the next command.