jneilliii/OctoPrint-BedLevelVisualizer #624: Adjustment Angle Data

rightonthedot, 2023-07-17:
**Is your feature request related to a problem? Please describe.**

The Mesh Z Height and Adjustment Angle data table shows values for every 10mm-15mm, or to some auto-fit function. The actual measured data is from a grid of points, so a bulk of that data is interpolated and not very helpful (which number is actually look at??)

**Describe the solution you'd like**

1 - Display values only at points where actual data was measured. The XY values should be the same as the points that are physically measured.... 

2 - Allow the user to define where their bed adjustments are located, and give interpolated dated for those specific points.

The first solution is great for determining the overall quality of the bed, the second solution is helpful to tramming operations. This is in context of an Ender3.

**Describe alternatives you've considered**

I try to find which value is closest to my actual adjustment points and go from there. 

**Additional context**

[image]

jneilliii, 2023-07-19:
IIRC the current RC version has a reference plane option that will calculate averages of the quadrants and display a 4 point plane in the graph. That is close to what your asking for.

rightonthedot, 2023-07-21:
I'll check it out! Thank you!

closed 2023-08-13 by github-actions[bot]