jneilliii/OctoPrint-BedLevelVisualizer #625: [FR]:  Change sizes of axis to allow setting up to actual bedsize?

Dazzlermc, 2023-07-22:
**Is your feature request related to a problem? Please describe.**
When performing a bed level the graph shows axis bigger than my actual printer bed , which causes the reading to be incorrect. 

**Describe the solution you'd like**
Allow a bed size to be set for the actual size of the bed measurements..

**Describe alternatives you've considered**
Tried changing the bed size in Octoprint , but no effect. 

**Additional context**

[image]

jneilliii, 2023-07-22:
The plugin pulls the bed size directly from the current printer profile dimensions in OctoPrint's settings. Once that is correct the next collection will update.

Dazzlermc, 2023-07-22:
It hasn’t pulled them down from my printer profile. Which is set to 220 x 235 x 250

jneilliii, 2023-07-22:
please enable debut logging in the plugin's settings, restart OctoPrint and try the update mesh button again. share your plugin_deblevelvisualizer_debug.log file from OctoPrint's logging section. You can drag it directly into a comment here in GitHub, you can't email it in a reply...

Dazzlermc, 2023-07-22:
[plugin_bedlevelvisualizer_debug (1).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/12137196/plugin_bedlevelvisualizer_debug.1.log)

jneilliii, 2023-07-22:
Thanks, can you share a screenshot of your active printer profile settings? The plugin is seeing 300x255x250 for some reason. 

> {'type': 'rectangular', 'x_min': 0, 'x_max': 300.0, 'y_min': 20, 'y_max': 255.0, 'z_min': 0, 'z_max': 250.0}

Dazzlermc, 2023-07-23:
[image]

jneilliii, 2023-07-23:
I'm very confused about this. could you do me one more favor, enable serial logging in OctoPrint's Serial Connection settings, do an update mesh process and then disable it back and share serial.log from OctoPrint's logging section? Also, to verify the printer profile above is the only printer profile you have or is the one that has the star next to it?

[image]

Dazzlermc, 2023-07-23:
That's not my printer profiles. ?

Dazzlermc, 2023-07-23:
[image]

jneilliii, 2023-07-23:
Yeah, that was a screenshot from my OctoPrint. Just to be 100% certain, what is the profile selected in the connection tab? 

[image]

Dazzlermc, 2023-07-23:
Ender3 is selected.
I have tried the other profiles and there is no difference.
[plugin_bedlevelvisualizer_debug (1).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/12139112/plugin_bedlevelvisualizer_debug.1.log)

jneilliii, 2023-07-23:
The only other reasons the size wouldn't match the current printer profile would be if it's being extracted from the mesh report from your printer, or you have some offsets configured in the plugin's settings. Do you have any offsets configured in the plugin's settings? If not, then I will definitely need the serial.log during the update mesh process.

Dazzlermc, 2023-07-23:
[serial (3).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/12139197/serial.3.log)

jneilliii, 2023-07-23:
Yeah, so it's definitely not interpreting the dimensions from the collection process. And if nothing is entered here in these settings it should be coming from the printer profile. 

[image]

Does it do the same thing if you turn off the reference plane option in visualization settings?

Dazzlermc, 2023-07-23:
offsets were not 0. I've set them at 0 and it is now reporting the size of the bed as correct. Graph still awful though, but I get a fresh start.  Thank you for your help.

jneilliii, 2023-07-23:
I think there was a bug in pulling the bed dimensions from the profile data anyway. I've refactored the reference plane code to use the min/max x/y values being passed by the collection process so they will now match up even with offsets applied. Will be available as version 1.1.2rc6 shortly.

closed 2023-07-23 by jneilliii