jneilliii/OctoPrint-BedLevelVisualizer #628: [BUG]:

Sandrock431, 2023-09-02:
**Describe the bug**
Unable to load existing Klipper bed mesh into visualizer (instead of having to run the mesh calibration each time).

**Expected behavior**
Using only the configuration below should be enough to load an existing Klipper bed mesh into the visualizer:
```
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
BED_MESH_OUTPUT		; report the bed leveling mesh points.
```

**Debug Log**
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/12503313/plugin_bedlevelvisualizer_debug.log)

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]

**Firmware and Version**
FIRMWARE_VERSION:v0.10.0-913-g5f990f93 FIRMWARE_NAME:Klipper

jneilliii, 2023-09-02:
so, I don't know anything about how klipper works and based on your debug log the plugin is not receiving any mesh data from klipper whatsoever. If you send the command `BED_MESH_OUTPUT` in the terminal tab what is returned?

Sandrock431, 2023-09-02:
Hmm. It looks like the Klipper behavior changed while I was away. It used to load the mesh by default, but it doesn't now. I'm able to create the behavior I expected with the sample below. It might be useful to update the visualizer documentation to include a snippet about loading the already-probed mesh:
```
BED_MESH_PROFILE LOAD=default
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
BED_MESH_OUTPUT		; report the bed leveling mesh points.
```

Thanks for hearing me out. I'll close this thread. P.S. Love the plugin!

closed 2023-09-02 by Sandrock431