jneilliii/OctoPrint-BedLevelVisualizer #629: [FR]: Better PSU Control support

0rphon, 2023-09-11:
Right now, if you want to start a mesh update you need to be connected to the printer already. If you aren't then the mesh update button is greyed out. 

This is a small thing, but it would be great if the plugin supported starting a mesh update without a connection if you have PSU Control set up, similar to the way that normal prints or the cura octoprint-connection plugin does it. That way the user doesn't need to manually turn the printer on before they can start a mesh update.

jneilliii, 2023-09-11:
This is a bit out of scope for what the plugin does when it's at simple as using a button in the navbar.

closed 2023-09-11 by jneilliii

0rphon, 2023-09-12:
all i was asking for is to be able to start a mesh update without an active connection so users could make use of the M80 command in their scripts. i don't see how adding that is out of scope, but i guess I'm not the dev here so *shrug*

jneilliii, 2023-09-12:
It's a little more complicated then that, you can't just send an M80 command if there's no printer connection. It would have to be via API or system command that you are able to turn the power on, then add a delay to give the printer time to power on and connect, and then continue with the remainder of the commands in the script. I would consider making the disabling of the update mesh button an option, but even then I don't think it's going to help in your case unless you have backpowering (not always a good thing, especially with Creality printers) of the printer, and in that case, the M80 command would already work.