jneilliii/OctoPrint-BedLevelVisualizer #631: [FR]:

DJDC13, 2023-10-06:
**Is your feature request related to a problem? Please describe.**
It is a little difficult to see the blue text when in dark mode (the blue on the dark background is not well defined.

**Describe the solution you'd like**
The option to customise the colours of the positive and negative offset values (the anti clockwise and clockwise rotation prompts and also the offsets in mm)

**Additional context**
I currently use Octoprint on a PC screen (so full sized 1440p screen) in dark mode.

jneilliii, 2023-10-07:
you can fix this by applying the desired css styling to your theming plugin. for UI Customizer use the advanced settings and add the style you want to override, reference [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/index.md#theme-support)

DJDC13, 2023-10-07:
Thanks. I'm not sure what any of that means but I'm assuming I need to get a plugin called UI Customizer?

jneilliii, 2023-10-07:
How are you getting dark mode if not using a plugin like UI customizer or themeify?

DJDC13, 2023-10-07:
yes I have themeify, I could not see any settings in there to change the bed visualiser stuff though

jneilliii, 2023-10-07:
Cool, so UI Customizer is very similar plus a whole lot more than Themify, but you can still fix this with Themeify I think. In the advanced options you add your own custom styles. As an example, let's take this style:

```
.table thead tr.info .raise, .table tbody tr .raise, .table tfoot tr.info .raise, .table tfoot tr td div span.raise {
    color: blue;
}
```

in Themeify, the selector would be the first part
```
.table thead tr.info .raise, .table tbody tr .raise, .table tfoot tr.info .raise, .table tfoot tr td div span.raise
```
the css-rule would be the next part
```
color
```
and then use the color picker to select the color you want to use.

closed 2023-10-08 by jneilliii