jneilliii/OctoPrint-BedLevelVisualizer #632: [BUG]: OctoDash web page has  CSRF validation errors

jneilliii, 2023-10-08:
**Describe the bug**
When using the `[!WEB]http://127.0.0.1/plugin/bedlevelvisualizer/bedlevelvisualizer` address for custom action in OctoDash the update mesh button no longer works. 

**Expected behavior**
To send the commands configured in plugin's settings

**Debug Log**
n/a

**Screenshots**
[image]

[image]

**Firmware and Version**
n/a

jneilliii, 2023-10-08:
Actually related to missing api key. added key to update mesh request. Brought up in https://github.com/UnchartedBull/OctoDash/issues/2887. Added in above commit. releasing v1.1.2rc7 to address.

atax112, 2023-10-08:
Thank you for the quick response. I am currently running a print, but will update ASAP

I just replace the files right? Never installed or updated anything outside of the UI or CLI
[image]

jneilliii, 2023-10-08:
Easiest method is go into OctoPrint's software update settings and change the release channel for the plugin to release candidate. You will be pumped to upgrade.

atax112, 2023-10-08:
Alright, I assume I cant see the plugins in the software update page since Im running a print and they are not manageable during jobs. Will check again after its done printing in a couple of hours.

[image]

jneilliii, 2023-10-08:
I'm not sure on that front, but if you don't see anything other than OctoPrint in that list after the print is complete then there's something else breaking the software update list. I've seen where on safari browsers you might not see the scroll bar, if that's the case for you. 

[image]

atax112, 2023-10-08:
Hmm, running on brave/chromium, will investigate after the print is done. 3.5h left but that will also be the middle of the night. I'll get back to you in the morning. Thanks a bunch.

atax112, 2023-10-09:
Ok, so no plugins on the software update page...strange.

So can I replace the files as mentioned before for now? I'll figure out the update page issue later.

atax112, 2023-10-09:
Any way to move the variance data back in its original place and make it visible again? For me it is obscure with a grey background on the left now, not readable in the  UI

[image]

jneilliii, 2023-10-09:
Check settings, I think there's options for where to place it. If you can't find it please open separate issue for that.

closed 2023-10-09 by jneilliii

Bond-Addict, 2025-02-04:
Just tried to do this myself. OctoDash asks me if I want to execute the command, but then doesnt do anything.

Bond-Addict, 2025-02-04:
[image]

[image]

jneilliii, 2025-02-04:
There's a bug in OctoDash that webcam embedded iframes don't load if you open from the main panel. If you swipe the notification panel down and click the button from that view it should load.