jneilliii/OctoPrint-BedLevelVisualizer #633: [BUG]: Mesh statistics not readable since rc7 update

atax112, 2023-10-09:
**Describe the bug**
After updating to rc7 to fix #632 mesh statistics are not readable by default. Background of the page section is opaque and position is to the left. No option in settings to move or change style from what Ive checked, only to enable/disable statistics.

[image]

When inspecting the elements in the browser I was able to modify css to get a transparent background but that is not a permanent fix.
[image]

**Expected behavior**
Data was shown without opaque background on right side of plotted data, expected the same after update.

**Firmware and Version**
1.1.2rc7

atax112, 2023-10-09:
UI customizer, Discoranged theme, tried different but didnt help
[image]

jneilliii, 2023-10-09:
Ok, thanks. That's the same I use (I ported that theme to UI customizer) and don't remember seeing the same symptom. I'll investigate further.

atax112, 2023-10-09:
Yeah I figured it wont be that, been using the theme with BLV 1.1.1 as well and it was ok...

jneilliii, 2023-10-09:
Might need some more details. Please share your settings for both UI Customizer and BLV plugins. I just installed on my work machine and am not reproducing the same thing as you are seeing. 

[image]

atax112, 2023-10-09:
Found it, its somehow browser related...
Chrome on the left is fine, although the readout is still on the left instead of right as before.

In brave on the right there has to be some dark theme, inverting whatever going on which isnt 100%
[image]

jneilliii, 2023-10-09:
> Chrome on the left is fine, although the readout is still on the left instead of right as before.

Yeah, I made that change for a reason, related to the addition of UBL data in 1.1.2rc1. With the block on the right it got too tall and was causing issues. This might be able to be overwritten with the new advanced section added in 1.1.2rc4. I'll test and see if I can provide a solution for that. 

It's odd that brave would work fine before, the rendering of the graph didn't change much outside of the positioning of those blocks.

jneilliii, 2023-10-09:
Unfortunately, it doesn't looks like the advanced settings are going to cut it for this because the annotation's text is being dynamically generated. I'll look into make the position configurable in the future.

atax112, 2023-10-09:
Oh I see.

I think there are other, more important features to put effort into, I dont mind it on the left, just thought that also something broke the position.

up to you at this point if you find the time.

thank you for the help

closed 2023-10-09 by atax112