jneilliii/OctoPrint-BedLevelVisualizer #639: [BUG]: Rendering Mesh display error

Sleeepy2, 2023-11-07:
While using RC 1.1.2rc7 switching to the BLV tab pops up and error in chrome and edge.
No issues running the level just switching to the tab.

Using OctoPrint 1.9.3 Python 3.9.2 OctoPi* 1.0.0cam (build 2023.07.20.144556)

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/13288992/plugin_bedlevelvisualizer_debug.log)

[image]

[image]

[image]

[image]

jneilliii, 2023-11-08:
I think I'll need one more bit of information @Sleeepy2 as I couldn't reproduce with just the points and got this result.

[image]

Could you enable serial logging for me and run through the process and then disable it and upload your serial.log please?

jneilliii, 2023-11-08:
Oh, and another thing. Could you provide your octoprint.log too so I can get the full list of the plugins you have installed. It's possible there is a plotly version issue if that's bundled in another plugin (ie Tasmota, TPLink PlotlyTempGraph).

Sleeepy2, 2023-11-08:
The Tasmota RC plugin fixed the issue. Looks like you were right and it's a plotly versions issue.

closed 2023-11-08 by Sleeepy2