jneilliii/OctoPrint-BedLevelVisualizer #643: [BUG]:

SectorOMEGAA, 2023-12-19:
Prusa Mini cannot display any visualization after probing across the bed.
[image]
Fresh octoprint and plugin install from the morning,  as I thought this might be the problem.
gcode used for the plugin:

```
M104 S170		; set extruder temp for bed leveling
M140 S60		; set bed temp
M109 R170		; wait for bed leveling temp
M190 S60		; wait for bed temp
G28			; home all without mesh bed level
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
G29			; mesh bed leveling
M104 S0			; cool down head
M140 S0			; cooling down the bed
```

Octoprint just timesout and does not return anything

jneilliii, 2023-12-19:
What shows in the terminal tab as the process starts? Usually this means that your firmware just doesn't report the mesh data back followed by an OK with the G29 command.

cealigbe, 2023-12-20:
I am having the same issue too on my Prusa Mini. I am running Firmware version 5.1.2 on my printer. My Octoprint is 1.9.3. Here is the log of what happens when I run the bed visualizer:

`Recv:  T:138.29/170.00 B:40.02/60.00 A:34.71/0.00 @:103 B@:127 W:?
Recv:  T:140.38/170.00 B:40.40/60.00 A:34.71/0.00 @:103 B@:127
Recv:  T:140.77/170.00 B:40.33/60.00 A:34.75/0.00 @:103 B@:127 W:?
Recv: echo:busy: processing

Recv:  T:170.07/170.00 B:58.93/60.00 A:35.24/0.00 @:30 B@:58
Recv:  T:170.05/170.00 B:58.93/60.00 A:35.15/0.00 @:30 B@:59 W:?
Recv:  T:170.07/170.00 B:58.95/60.00 A:35.28/0.00 @:30 B@:62 W:?
Recv: echo:busy: processing
Recv:  T:170.14/170.00 B:58.95/60.00 A:35.22/0.00 @:30 B@:66
Recv:  T:170.21/170.00 B:58.98/60.00 A:35.20/0.00 @:29 B@:63 W:?
Recv:  T:170.21/170.00 B:58.98/60.00 A:35.24/0.00 @:29 B@:67 W:?
Recv: echo:busy: processing
Recv:  T:170.23/170.00 B:59.02/60.00 A:35.20/0.00 @:29 B@:66
Recv:  T:170.19/170.00 B:59.04/60.00 A:35.20/0.00 @:30 B@:65 W:4
Recv:  T:170.19/170.00 B:59.05/60.00 A:35.22/0.00 @:30 B@:67 W:3
Recv: echo:busy: processing
Recv:  T:170.23/170.00 B:59.08/60.00 A:35.20/0.00 @:29 B@:67
Recv:  T:170.12/170.00 B:59.11/60.00 A:35.24/0.00 @:30 B@:64 W:2
Recv:  T:170.16/170.00 B:59.18/60.00 A:35.22/0.00 @:30 B@:60 W:1
Recv: echo:busy: processing
Recv:  T:170.16/170.00 B:59.19/60.00 A:35.22/0.00 @:30 B@:61
Recv:  T:170.19/170.00 B:59.21/60.00 A:35.26/0.00 @:30 B@:60 W:0
Recv: ok
Send: G28
Recv: echo:endstops hit:  Z:1.99
Recv:  T:170.09/170.00 B:59.35/60.00 A:35.17/0.00 @:30 B@:50
Recv: echo:busy: processing
Recv:  T:170.16/170.00 B:59.52/60.00 A:35.22/0.00 @:30 B@:38
Recv: echo:endstops hit:  X:147.12
Recv: echo:busy: processing
Recv: echo:endstops hit:  Y:-14.72
Recv:  T:170.21/170.00 B:59.62/60.00 A:35.13/0.00 @:29 B@:33
Recv: echo:endstops hit:  Z:-2.63
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:-2.01
Recv:  T:169.93/170.00 B:59.67/60.00 A:35.15/0.00 @:31 B@:34
Recv: X:176.40 Y:24.10 Z:2.00 E:0.00 Count X:17640 Y:2410 Z:800
Recv: ok
Send: G29
Recv: Default storage slot 0 selected.
Recv: Mesh invalidated. Probing mesh.
Recv:  T:169.63/170.00 B:59.70/60.00 A:35.13/0.00 @:33 B@:36
Recv: echo:busy: processing
Recv:  T:169.65/170.00 B:59.69/60.00 A:35.06/0.00 @:33 B@:42
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.60
Recv:  T:169.50/170.00 B:59.70/60.00 A:35.06/0.00 @:34 B@:44
Recv: echo:endstops hit:  Z:0.60
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.60
Recv:  T:169.54/170.00 B:59.70/60.00 A:35.09/0.00 @:33 B@:46
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.93
Recv: echo:endstops hit:  Z:0.91
Recv:  T:169.61/170.00 B:59.71/60.00 A:35.07/0.00 @:33 B@:46
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.18
Recv: echo:endstops hit:  Z:1.17
Recv: echo:endstops hit:  Z:1.15
Recv:  T:169.78/170.00 B:59.75/60.00 A:35.04/0.00 @:32 B@:44
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.35
Recv: echo:endstops hit:  Z:1.33
Recv:  T:169.96/170.00 B:59.75/60.00 A:35.02/0.00 @:31 B@:46
Recv: echo:endstops hit:  Z:1.33
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.39
Recv:  T:169.96/170.00 B:59.83/60.00 A:35.02/0.00 @:31 B@:38
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.38
Recv: echo:endstops hit:  Z:1.38
Recv:  T:169.93/170.00 B:59.83/60.00 A:34.96/0.00 @:31 B@:40
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.24
Recv: echo:endstops hit:  Z:1.23
Recv: echo:endstops hit:  Z:1.22
Recv:  T:170.00/170.00 B:59.84/60.00 A:35.02/0.00 @:31 B@:40
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.01
Recv: echo:endstops hit:  Z:1.01
Recv:  T:170.00/170.00 B:59.86/60.00 A:34.98/0.00 @:31 B@:40
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.00
Recv: echo:endstops hit:  Z:0.71
Recv:  T:170.09/170.00 B:59.90/60.00 A:34.96/0.00 @:31 B@:37
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.70
Recv: echo:endstops hit:  Z:0.70
Recv:  T:170.25/170.00 B:59.91/60.00 A:35.00/0.00 @:29 B@:37
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.66
Recv: echo:endstops hit:  Z:0.65
Recv:  T:170.25/170.00 B:59.95/60.00 A:35.02/0.00 @:30 B@:34
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.98
Recv: echo:endstops hit:  Z:0.97
Recv:  T:170.28/170.00 B:59.93/60.00 A:35.04/0.00 @:30 B@:36
Recv: echo:endstops hit:  Z:0.97
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.12
Recv:  T:170.42/170.00 B:59.88/60.00 A:35.00/0.00 @:29 B@:43
Recv: echo:endstops hit:  Z:1.12
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.11
Recv:  T:170.51/170.00 B:59.90/60.00 A:34.90/0.00 @:28 B@:42
Recv: echo:endstops hit:  Z:1.23
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.22
Recv: echo:endstops hit:  Z:1.21
Recv:  T:170.37/170.00 B:59.91/60.00 A:34.96/0.00 @:29 B@:41
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.03
Recv: echo:endstops hit:  Z:1.02
Recv:  T:169.93/170.00 B:59.93/60.00 A:34.98/0.00 @:32 B@:38
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.92
Recv:  T:169.87/170.00 B:59.95/60.00 A:34.98/0.00 @:32 B@:37
Recv: echo:endstops hit:  Z:0.92
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.91
Recv:  T:169.80/170.00 B:59.95/60.00 A:34.96/0.00 @:32 B@:37
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.79
Recv: echo:endstops hit:  Z:0.79
Recv: echo:endstops hit:  Z:0.78
Recv:  T:169.74/170.00 B:59.99/60.00 A:34.96/0.00 @:32 B@:33
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:0.55
Recv: echo:endstops hit:  Z:0.54
Recv:  T:169.74/170.00 B:59.95/60.00 A:34.96/0.00 @:32 B@:39
Recv: echo:endstops hit:  Z:0.54
Recv: echo:busy: processing
Recv: X:35.20 Y:9.80 Z:2.00 E:0.00 Count X:3520 Y:980 Z:800
Recv: Extrapolating mesh...done
Recv: Unified Bed Leveling System v1.01 active
Recv: ok
Send: M104 S0
Recv: ok
Send: M140 S0
Recv: ok
[image]

Recv:  T:169.89/0.00 B:59.92/0.00 A:34.83/0.00 @:0 B@:0
Recv:  T:169.87/0.00 B:59.91/0.00 A:34.79/0.00 @:0 B@:0`

[image]

jneilliii, 2023-12-21:
> Recv: echo:busy: processing
Recv: X:35.20 Y:9.80 Z:2.00 E:0.00 Count X:3520 Y:980 Z:800
Recv: Extrapolating mesh...done
Recv: Unified Bed Leveling System v1.01 active
Recv: ok
Send: M104 S0
Recv: ok
Send: M140 S0
Recv: ok

Yeah, so no mesh data is actually returned by your firmware. Is there any chance there's a different command for G29 that will report the mesh?

evie-lau, 2023-12-23:
I ran into this too with logs looking the same.
I was able to get the bed topography report by changing G29 to `G29 T` as mentioned in the [README](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/README.md)
I also surrounded the G29 command with M155 S30 and M155 S3 as noted, but not sure if that made any difference.

jneilliii, 2023-12-23:
Great, so basically the same as standard Marlin bilinear.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md#bilinear-bed-levelling

Defozo, 2023-12-23:
```
M104 S170		; set extruder temp for bed leveling
M140 S60		; set bed temp
M109 R170		; wait for bed leveling temp
M190 S60		; wait for bed temp
G28			; home all without mesh bed level
M155 S30
G29
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
G29 T			; mesh bed leveling
M155 S3
M104 S0			; cool down head
M140 S0			; cooling down the bed
```

fevofevo, 2023-12-29:
Hi guys, great, tried many ways but this actually works :)

But how are the correction values to be interpreted? Do you use the outer values? At least the in-between screws have to be averaged, right? Which option did you choose that actually works in the end?

[image]

jneilliii, 2023-12-29:
personally I've never really liked the screw adjustment UI because of this very reason. It doesn't make sense unless the probe points are directly over the screw locations. I would try to pick the point that is close to the screw location and use that as the adjustment.

closed 2024-01-21 by github-actions[bot]