jneilliii/OctoPrint-BedLevelVisualizer #645: [FR]: Parse/Display Z-offset from ABL process.

AJolly, 2023-12-21:
Some printers like anycubic kobra 2's have a way to automatically measure the Z offset as part of their abl process:
It would be great If the plugin could pull that info in as well.

Send: G29
Recv: Preheating
[...]
Recv: measured_z: -3.23
Recv: measured_z: 0.12
Recv: measured_z: 0.27
Recv: measured_z: 0.35

p.s - Is there a way to increase the precision?  I notice that the plugin parses out the info from a M420, but M503 gives me the same info with more precision:
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.115 +0.270 +0.346 +0.358 +0.322
Recv:  1 +0.240 +0.345 +0.376 +0.347 +0.294
Recv:  2 +0.228 +0.308 +0.320 +0.278 +0.222
Recv:  3 +0.088 +0.185 +0.205 +0.173 +0.115
Recv:  4 -0.065 +0.011 +0.051 -0.012 -0.090

Send: M503
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 ; Leveling ON
Recv: echo:  G29 W I0 J0 Z0.11512
Recv: echo:  G29 W I1 J0 Z0.27012
Recv: echo:  G29 W I2 J0 Z0.34562
Recv: echo:  G29 W I3 J0 Z0.35762
Recv: echo:  G29 W I4 J0 Z0.32162
Recv: echo:  G29 W I0 J1 Z0.24012
Recv: echo:  G29 W I1 J1 Z0.34512
Recv: echo:  G29 W I2 J1 Z0.37612
Recv: echo:  G29 W I3 J1 Z0.34712
Recv: echo:  G29 W I4 J1 Z0.29412
Recv: echo:  G29 W I0 J2 Z0.22812
Recv: echo:  G29 W I1 J2 Z0.30812
Recv: echo:  G29 W I2 J2 Z0.32012
Recv: echo:  G29 W I3 J2 Z0.27812
Recv: echo:  G29 W I4 J2 Z0.22162
Recv: echo:  G29 W I0 J3 Z0.08812
Recv: echo:  G29 W I1 J3 Z0.18512
Recv: echo:  G29 W I2 J3 Z0.20462
Recv: echo:  G29 W I3 J3 Z0.17312
Recv: echo:  G29 W I4 J3 Z0.11462
Recv: echo:  G29 W I0 J4 Z-0.06488
Recv: echo:  G29 W I1 J4 Z0.01062
Recv: echo:  G29 W I2 J4 Z0.05062
Recv: echo:  G29 W I3 J4 Z-0.01238
Recv: echo:  G29 W I4 J4 Z-0.09038

jneilliii, 2023-12-21:
> Send: G29
> Recv: Preheating
> [...]
> Recv: measured_z: -3.23
> Recv: measured_z: 0.12
> Recv: measured_z: 0.27
> Recv: measured_z: 0.35

This would be tricky if it reports more than one, what comes right after this last line? 

> p.s - Is there a way to increase the precision? I notice that the plugin parses out the info from a M420, but M503 gives me the same info with more precision:

probably not without upgrading firmware, I suspect that is something baked in and would have to be set, compiled, and flashed. at least, I'm not aware of anyway to increase precision.

jneilliii, 2023-12-21:
> Recv: measured_z: -3.23
> Recv: measured_z: 0.12
> Recv: measured_z: 0.27
> Recv: measured_z: 0.35

actually, these are the individual probe points, being reported for each spot. the overall report at the end is just those in table format, so the plugin is getting these already...

closed 2024-01-12 by github-actions[bot]