jneilliii/OctoPrint-BedLevelVisualizer #646: [BUG]:

KevinWGarrett, 2023-12-25:
**Describe the bug**

It no longer collects bed level data. It never returns a mesh to visualize. I have to cancel eventually.
I think it must not be compatible with Prusa Firmware 5.1.0, which starts the bed leveling routine at the back left corner instead of the front right corner like it used to.

**Expected behavior**

I expect it to work like it used to. I click "Update Mesh Now", the printer goes through the bed leveling routine, and then the mesh is graphically displayed. Now it never returns a mesh.

**Debug Log**

[2023-12-24 15:38:36,172] DEBUG: mesh collection started
[2023-12-24 15:40:36,714] DEBUG: mesh collection started
[2023-12-24 15:52:11,001] DEBUG: Canceling mesh collection per user request
[2023-12-24 15:52:11,002] DEBUG: Mesh data collected prior to cancel:
[2023-12-24 15:52:11,002] DEBUG: []
[2023-12-24 15:52:11,003] DEBUG: Mesh data after clearing:
[2023-12-24 15:52:11,003] DEBUG: []

Notice that I waited about 12 minutes before cancelling. 

The GCODE I'm using is for the Prusa Mini (which is what I have). I have also tried the default code:

"M104 S170		; set extruder temp for bed leveling
M140 S60		; set bed temp
M109 R170		; wait for bed leveling temp
M190 S60		; wait for bed temp
G28			; home all without mesh bed level
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
G29			; mesh bed leveling
G81			; report mesh bed leveling status.
M104 S0			; cool down head
M140 S0			; cooling down the bed"

**Screenshots**

**Firmware and Version**

Prusa Firmware 5.1.0
OctoPrint 1.9.3 
Python 3.7.3 
OctoPi 0.18.0
[image]
[image]
[image]
[image]
[image]

KevinWGarrett, 2023-12-25:
I tend to think now that the issue is with the new firmware, not your code. I think it is simply not returning a mesh.

closed 2023-12-25 by KevinWGarrett

jneilliii, 2023-12-25:
Try this....
```
G29
@BEDLEVELVISUALIZER
G29 T
```

KevinWGarrett, 2023-12-26:
Well, indeed that did work. Which confuses me. The results I was getting from First Layer Calibration made me think that the mesh was not being returned. At first, I thought maybe my bed had gotten so out of level that the mesh correction was not sufficient to correct the printing. That's why I turned to your plug-in.

I don't know how it got so out of level. I had recently leveled the bed. I have the silicone bed leveling mod installed on my Mini and had dialed it in to near perfection only recently.

BTW, another question. The Mini has 8 adjustment screws. It's a 3 x 3 matrix with the center being fixed. When I first used your plugin to level the bed after installing the silicone mod, I swear it showed the correct number of adjustments. Now I cannot get it to show only 9 points. Now it shows a 6 x 6 matrix. I've tried various settings but cannot get it to show a 3 x 3 screw adjustment table.

Thank you for your help. Your plugin is really useful.

P.S. While searching the web for instructions on how to get your plugin to show the correct number of adjustments, I discovered the Prusa Leveling Guide Plugin and installed it. It likewise does not return (or begin probing for) a mesh. :) Yeah, something changed alright with firmware 5.1.x.

Regards,

Kevin Garrett

[Screenshot 2023-12-26 at 8.10.24 AM.png]

KevinWGarrett, 2023-12-26:
Dang it. I knew there was something else I wanted to show you. Here are photos of First Layer Calibration. Note the horrible results on the ride side of the bed. The printer's leveling routine did not correct for it for the smooth sheet. For the textured sheet, it came out OK. Last photo shows that the bed level is too high on the right size as compared to the left during a print (and low in the middle?)

Kevin Garrett

[IMG_6929-50%.jpeg]

[IMG_6930-50%.jpeg][IMG_6935-50%.jpeg]