jneilliii/OctoPrint-BedLevelVisualizer #650: [FR]: Manual grid visualization

Elarion245, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
The feature request is related to the missing Lerdge Z support and describes a simple generic solution for unsupported printers.

**Describe the solution you'd like**
Unsupported printers don't follow the standard terminal output (see e.g. https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/167). To still use the plugin, irrespectively of the output, it would be great to allow a visualization of manually filled z-values.
Inputs could be:
- Activation of manual mode field
- Size of grid
- Input fields according to grid size
- Apply and cancel buttons

**Describe alternatives you've considered**
Asking the Lerdge team to support the G29 - won't come soon

**Additional context**
N/A

jneilliii, 2024-01-27:
I'll be more keen to support the lerdge output than do something like this. In the meantime, you could use a tool like this https://i.chillrain.com/index.php/3d-printer-auto-bed-leveling-mesh-visualizer/

closed 2024-01-27 by jneilliii

Elarion245, 2024-01-27:
Thank you for the link and your viewpoint. Obviously, I'd prefer the lerdge as well, but having a "last chance" mode sounded meaningfull as well