jneilliii/OctoPrint-BedLevelVisualizer #652: [BUG]: Prusa mini Timeout occured before processing completed.

beye91, 2024-02-25:
**Describe the bug**
Hello, I have a new setup and I never worked with the BedLevelVisualizer plugin. 

All settings are default. I added the GCODE from the wiki for the Prusa Mini. The processing timeout is 1800. I also tried with 3600 which did not change anything. 

I am getting the following error:
Timeout occured before processing completed. Processing may still be running or there may be a configuration error. Consider increasing the Processing Timeout value in settings and restart OctoPrint.

**Debug Log**
Terminal Log:
```
Send: M104 S170
Recv: ok
Send: M140 S60
Recv: ok
Send: M109 R170
Recv:  T:107.26/170.00 B:49.05/60.00 A:36.71/0.00 @:0 B@:0 W:?
Recv:  T:107.16/170.00 B:49.08/60.00 A:36.78/0.00 @:107 B@:127
Recv:  T:106.96/170.00 B:48.98/60.00 A:36.84/0.00 @:109 B@:127 W:?
Recv:  T:108.29/170.00 B:48.93/60.00 A:37.08/0.00 @:100 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:109.03/170.00 B:48.93/60.00 A:37.08/0.00 @:95 B@:127
Recv:  T:110.79/170.00 B:48.83/60.00 A:37.11/0.00 @:83 B@:127 W:?
Recv:  T:113.78/170.00 B:48.83/60.00 A:37.17/0.00 @:64 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:114.76/170.00 B:48.90/60.00 A:37.21/0.00 @:58 B@:127
Recv:  T:116.55/170.00 B:48.98/60.00 A:37.30/0.00 @:51 B@:127 W:?
Recv:  T:119.13/170.00 B:49.10/60.00 A:37.30/0.00 @:56 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:119.76/170.00 B:49.14/60.00 A:37.35/0.00 @:57 B@:127
Recv:  T:121.00/170.00 B:49.34/60.00 A:37.35/0.00 @:61 B@:127 W:?
Recv:  T:122.77/170.00 B:49.54/60.00 A:37.41/0.00 @:67 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:123.37/170.00 B:49.61/60.00 A:37.50/0.00 @:69 B@:127
Recv:  T:124.51/170.00 B:49.84/60.00 A:37.46/0.00 @:72 B@:127 W:?
Recv:  T:126.54/170.00 B:50.16/60.00 A:37.56/0.00 @:77 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:127.18/170.00 B:50.20/60.00 A:37.52/0.00 @:78 B@:127
Recv:  T:128.43/170.00 B:50.43/60.00 A:37.57/0.00 @:80 B@:127 W:?
Recv:  T:130.34/170.00 B:50.68/60.00 A:37.61/0.00 @:82 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:131.06/170.00 B:50.82/60.00 A:37.63/0.00 @:82 B@:127
Recv:  T:132.52/170.00 B:50.99/60.00 A:37.63/0.00 @:82 B@:127 W:?
Recv:  T:135.05/170.00 B:51.26/60.00 A:37.68/0.00 @:81 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:135.74/170.00 B:51.38/60.00 A:37.70/0.00 @:81 B@:127
Recv:  T:137.14/170.00 B:51.72/60.00 A:37.68/0.00 @:81 B@:127 W:?
Recv:  T:139.16/170.00 B:51.97/60.00 A:37.74/0.00 @:82 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:139.78/170.00 B:52.05/60.00 A:37.74/0.00 @:83 B@:127
Recv:  T:141.30/170.00 B:52.29/60.00 A:37.78/0.00 @:82 B@:127 W:?
Recv:  T:143.85/170.00 B:52.64/60.00 A:37.81/0.00 @:81 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:144.56/170.00 B:52.76/60.00 A:37.89/0.00 @:80 B@:127
Recv:  T:145.92/170.00 B:52.93/60.00 A:37.92/0.00 @:80 B@:127 W:?
Recv:  T:147.92/170.00 B:53.30/60.00 A:37.90/0.00 @:81 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:148.54/170.00 B:53.37/60.00 A:37.89/0.00 @:81 B@:127
Recv:  T:149.79/170.00 B:53.58/60.00 A:38.00/0.00 @:82 B@:127 W:?
Recv:  T:152.16/170.00 B:53.93/60.00 A:37.94/0.00 @:82 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:152.84/170.00 B:54.05/60.00 A:38.01/0.00 @:82 B@:127
Recv:  T:154.16/170.00 B:54.18/60.00 A:38.01/0.00 @:82 B@:127 W:?
Recv:  T:156.07/170.00 B:54.53/60.00 A:38.01/0.00 @:83 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:156.71/170.00 B:54.62/60.00 A:38.05/0.00 @:83 B@:127
Recv:  T:157.93/170.00 B:54.87/60.00 A:38.05/0.00 @:83 B@:127 W:?
Recv:  T:159.92/170.00 B:55.20/60.00 A:38.16/0.00 @:86 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:160.59/170.00 B:55.35/60.00 A:38.18/0.00 @:10 B@:127
[... 157 lines cut ...]
Recv: echo:busy: processing
Recv: echo:endstops hit:  Z:1.65
Recv: X:35.20 Y:9.80 Z:2.00 E:-6.80 Count X:3520 Y:980 Z:800
Recv: Extrapolating mesh...done
Recv: Unified Bed Leveling System v1.01 active
Recv: ok
Send: M104 S0
Recv: ok
Send: M140 S0
Recv: ok
```

plugin_bedlevelvisualizer_debug.log:
```
[2024-02-25 14:47:01,023] DEBUG: mesh collection started
[2024-02-25 14:48:00,012] DEBUG: mesh collection started
[2024-02-25 15:17:01,053] DEBUG: Canceling mesh collection per user request
[2024-02-25 15:17:01,055] DEBUG: Mesh data collected prior to cancel:
[2024-02-25 15:17:01,056] DEBUG: []
[2024-02-25 15:17:01,056] DEBUG: Mesh data after clearing:
[2024-02-25 15:17:01,056] DEBUG: []
```

**Firmware and Version**
Currently, I am running OctoPrint version 1.9.3 with Bed Visualizer (1.1.2rc10) on a RaspberryPi 3B.

beye91, 2024-02-25:
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  -41,226)                              (195,226)
Recv:         0       1       2       3       4       5
Recv:  5 | +0.100  +0.141  +0.225  +0.164  +0.077  -0.039
Recv:    |
Recv:  4 | +0.171  +0.234  +0.296  +0.210  +0.109  +0.007
Recv:    |
Recv:  3 | +0.285  +0.326  +0.367  +0.256  +0.140  +0.024
Recv:    |
Recv:  2 | +0.205  +0.265 [+0.325] +0.174  +0.100  +0.026
Recv:    |
Recv:  1 | +0.145  +0.192  +0.240  +0.147  -0.008  -0.163
Recv:    |
Recv:  0 | +0.060  +0.120  +0.155  +0.121  -0.115  -0.189
Recv:         0       1       2       3       4       5
Recv:     (  -41,  -48)                              (195,  -48)
Recv: 
Recv: ok

How does this output relate to the 12-bed leveling measurements? I tried to use G26 T in the GCODE but it will only measure one point on the bed out of 12. 

M420 V creates an image. 
[image]

jneilliii, 2024-02-25:
> If I understand correctly G29 T is generating the report and G29 is doing the measurement. Is that correct?

I have no idea how Prusa has implemented this in their firmware. On standard Marlin bilinear firmware when you run `G29 T` it will do the probing and generate the report at the end of the process that the plugin can use to graph. If that's true with the mini's firmware then you don't need to double them up and it would just be like this.

```
M104 S170		; set extruder temp for bed leveling
M140 S60		; set bed temp
M109 R170		; wait for bed leveling temp
M190 S60		; wait for bed temp
G28			; home all without mesh bed level
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
G29 T			; mesh bed leveling
M104 S0			; cool down head
M140 S0			; cooling down the bed
```

beye91, 2024-02-25:
That seems not to be the case, with G29 T the complete probing is not starting. So I think the combination of the above works for me just fine. :) Thanks for your help! Much appreciated.

closed 2024-02-25 by beye91