jneilliii/OctoPrint-BedLevelVisualizer #659: Gcode for Prusa MK3.9 / MK4

Kehet, 2024-03-25:
This is a problem with documentation. Currently, [the documentation](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md#prusa-firmware) suggests that Prusa would report status with `G81`. However, this seems to be now disabled and the printer responds with following error:

```
Recv: echo:Unknown command: "G81"
```

I found [this gist](https://gist.github.com/TheZackCodec/5545ff29c7988c1d3e3a7e8793a0a57a#file-mk4_bed_visualizer-gcode) which suggests that `G29 T` would print mesh like XL (and I can confirm this seems to be the case). 

Since the documentation only has gcode for Prusa XL5T, I suggest adding gcode for MK3.9 and MK4, which is virtually the same but without all this tool head swapping (like in that gist).

jneilliii, 2024-03-25:
That's because these examples are based on older versions of the MK printer. The Prusa XL example was contributed by another user. Happy to update the examples again with a MK3.9/4 specific example.

jneilliii, 2024-03-25:
examples updated

closed 2024-03-25 by jneilliii

timur-tabi, 2024-04-04:
I think this also applies to the MK3.5

timur-tabi, 2024-04-04:
MK3.5 upgraded from an MK3s.

Recv: FIRMWARE_NAME:Prusa-Firmware-Buddy 5.2.2+13941 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:5 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:1
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok

OopAck, 2024-05-31:
Is there GCode for Mk3.5 for the Bed Visualizer?

jneilliii, 2024-05-31:
try thus

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md#prusa-mk394

OopAck, 2024-05-31:
Thank you
Will try it out.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md#prusa-mk394