jneilliii/OctoPrint-BedLevelVisualizer #661: [BUG]:

Samtecx, 2024-03-29:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

Samtecx, 2024-03-29:
Can someone tell me how I can restore the correct number of setting values, I have far too many as shown here
[image]

jneilliii, 2024-03-29:
same issue as #637 and #638, your firmware is reporting that many points in its mesh report. in one of the other issues I commented on using either UI Customizer to widen the available area of the tab. Since this is a duplicate, I will be closing.

closed 2024-03-29 by jneilliii