jneilliii/OctoPrint-BedLevelVisualizer #665: [BUG]: The custom command icon picker does not have the search function enabled

sharky98, 2024-07-07:
**Describe the bug**
When using the icon picker, there is no search function, even though there should be according to the jinja template.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/4f460c743bf82fffec8ea5c407f92c050d3ba7a4/octoprint_bedlevelvisualizer/templates/bedlevelvisualizer_settings.jinja2#L221

**Expected behavior**
Search function active.

This may be due to the footer not being shown by default? IE, not having `showFooter: true`?

**Screenshots**
[image]

**Firmware and Version**
BedLevelVisualizer v1.1.1

jneilliii, 2024-07-07:
You're completely right. The options need to include `showFooter: true` for the search to show, however that introduces another bug, where when you click in the input box, it instantly closes the window and you can't type. I think I was actually using the searchInFooter and not showing the footer to hide the input box because of this. 

However, I was able to figure out I could define the container property, and tweak the popover template to get it to show in the middle of the tab with search functionality. 

[image]

I'll add this change and release a new rc version with it.

jneilliii, 2024-07-07:
1.1.2rc11 released with the above commit. You can change the plugin's release channel in Software Update settings to `Release Candidate` and upgrade when prompted.