jneilliii/OctoPrint-BedLevelVisualizer #666: [BUG]: Ender 3 Pro Bilinear bed levelling don't probe the points on the bed

lopio1234, 2024-08-12:
**Describe the bug**
The bl touch wont proble the probing points

**Expected behavior**
I use the gcode from the gcode-examples.md for Bilinear leveling, but my ender 3 pro just home all axes heats the bed and the calibration completed tones ringing

the probing with UBL work as intendet with the sample gcode

**Debug Log**
[plugin_bedlevelvisualizer_debug (1).log](https://github.com/user-attachments/files/16588988/plugin_bedlevelvisualizer_debug.1.log)
[ender3pro_terminal_blv_log.txt](https://github.com/user-attachments/files/16589026/ender3pro_terminal_blv_log.txt)

**Screenshots**
[image]
After homing xy and z

**Firmware and Version**
OctoPrint 1.10.2 Python 3.7.3 OctoPi 0.18.0
Bed Visualizer Version 1.1.2rc11
Marlin 2.1.2.1

jneilliii, 2024-08-12:
> the probing with UBL work as intendet with the sample gcode

if that works, then that's what your firmware is configured for and you should use those commands instead of bilinear.

lopio1234, 2024-08-12:
> > the probing with UBL work as intendet with the sample gcode
> 
> if that works, then that's what your firmware is configured for and you should use those commands instead of bilinear.

yeah, but "before" i'd used the Bilinear variante of probing. I guess i'm fine with the ubl probing, but i wonder why it suddently broke the Bilinear probing.

jneilliii, 2024-08-12:
It usually will only change if you upgrade/flash new firmware. I don't have this specific printer, so not really sure from that perspective of what it should be, but as long as you are getting it to probe and the plugin is mapping the returned results, then there's nothing really here to do so to say.

closed 2024-08-12 by jneilliii