jneilliii/OctoPrint-BedLevelVisualizer #673: [FR]:  Improved Display of Correction Angles and Screw Adjustment Calculations in OctoPrint Bed Visualizer

obilger, 2024-12-03:
**Is your feature request related to a problem? Please describe.**

Currently, the plugin provides a detailed visualization of the bed level and displays the correction angles in a table format with X=0 and Y=0 positioned at the top left, corresponding to the front-left corner of the bed. While this information is useful, I believe it would be much easier to interpret if these values were displayed directly on a top-down view of the bed, including the positions of the four leveling screws.

**Describe the solution you'd like**

Here are the specific features I am proposing:

Top-Down Bed View with Correction Angles: Display the correction angles on a visual representation of the bed as seen from above, with clear indications of the positions of the four leveling screws. This would provide a more intuitive understanding of the bed's tilt and necessary adjustments.
Screw Adjustment Calculations: In addition to displaying the correction angles, calculate and show the exact adjustments needed for each bed leveling screw. This would make the leveling process more straightforward and precise, helping users achieve a perfectly leveled bed more efficiently.

**Describe alternatives you've considered**

**Additional context**

jneilliii, 2024-12-03:
You can change the orientation of the data tables (including adjustments) using the options in the plugin, to make the 0,0 point to be bottom left. 

[image]

You can also click on any point in the table to make that the "0" z height and the numbers will change. That being said, I've never really been happy with the adjustment table (was contributed by someone else) and never use it. I use my bed leveling wizard plugin or the autobim plugin for getting the adjustments done. In fact, in the current RC versions I was working on removing the adjustment table and putting those turn directions/amounts into the probe point hover details (still working on it as you can tell). 

[image]

You'll notice that not just the angles are shown on the table. When you hover your mouse over the table cells or in the rc version, the points on the mesh, it shows the number of turns on the right, not just the angle.

jneilliii, 2024-12-03:
It's on the data tab, not the corrections tab. That setting will effect both tables.

closed 2024-12-26 by github-actions[bot]