jneilliii/OctoPrint-BedLevelVisualizer #678: Prusa Corrections and the Prusa Mini

reox, 2025-03-02:
I tried to level the bed using the "Show Prusa Correction Values" but these values do not really make sense to me. I found out I need to switch the values for back&front and left&right to get a reasonable result - but flipping the axes in the software gives really different results?
I have the feeling everything is inverted - or I did something utterly wrong... Did I setup the printer incorrectly?
And another question: Should you also enable "Use Relative Z Offsets"?

jneilliii, 2025-03-02:
It does sound like flip axes might be required if the values are inverted like you think but I'm not even sure if those values are valid for prusa buddy firmware, it was a feature designed for the older mks firmware and may not be valid for you.

I'd recommend putting something in the front right corner that will be picked up by the probing prices and running an update mesh process. See in the graph where the bump up shows and adjust flip axes options as necessary.

reox, 2025-03-03:
Will try that!

I think the ordering of the probe-points changed between firmwares? In the new firmware the printer starts in the upper left corner, while the old firmware started in the lower right, IIRC.

jneilliii, 2025-03-03:
> I think the ordering of the probe-points changed between firmwares?

I have no idea, I don't have either of those printers.