jneilliii/OctoPrint-BedLevelVisualizer #679: [FR]: UBL or a larger mesh with more grid points

jcase16, 2025-03-11:
**Is your feature request related to a problem? Please describe.**
I have an Ender3 with a 400X400 glass bed and I'm really starting to become aware that the 12 grid points used in the standard bed leveling is just not enough for my bed.  If I level the bed with your tool and print something right on one of the points then it generally prints great.  If I print away from them then I'm much more likely to have first layer squish issues in parts.  I would love to be able to use UBL (still learning about this and I may be misinformed) so that I can visualize the bed with 36, 64 or even 100 grid points.  Any chance that's possible?

**Describe the solution you'd like**
I would love to be able to use UBL (still learning about this and I may be misinformed) so that I can visualize the bed with 36, 64 or even 100 grid points.  Any chance that's possible?

**Describe alternatives you've considered**
Haven't gotten there yet.  Only alternative has been looking for other plugins that might offer this and so far I have found nothing.

**Additional context**

jneilliii, 2025-03-11:
Well, the plugin does struggle with larger number of probe points, but that doesn't impact if bed leveling is working properly or not. It's the firmware's job to handle the leveling procedure and adjust for the variance in your bed, not this plugin, so not really sure there's anything in this issue that hasn't already been brought up in #638 relative to more than a 10x10 grid.

closed 2025-03-11 by jneilliii

jcase16, 2025-03-12:
I actually figured out that it did work.  It was a 10X10 but something's wrong with my printer.  All of the grid points are 0's and when I watch it, it homes and then does nothing.  If I send G29 G to the Terminal, it does nothing.  Bah...  Thanks for your help.