jneilliii/OctoPrint-BedLevelVisualizer #680: [FR]: Historical data,

brodergloosy, 2025-04-09:
**Is your feature request related to a problem? Please describe.**

I'm frustrated when I forget to take a picture of my mesh graph/chart before creating a new mesh. I'm also frustrated that as I was writing this, I was looking at the plug in settings, and I discovered that there was a "save snapshot" button in the settings was disabled by default. I get it, it's my fault for not checking through the settings thoroughly upon install, but it would be nice if it was Enabled by default so it doesn't happen to others. 

**Describe the solution you'd like**

It would be nice if there was a way to view stored data for the previous ~3 meshes recorded side by side to compare

**Describe alternatives you've considered**

I've considered not dropping things on my printer, bumping the frame, or resisting the urge to tinker with something innocuous that inevitably throws it all out of wack LOL. But when this happens and I'm in a hurry to print, I always forget to take a screenshot of the previous mesh data so I can compare the changes and understand.

**Additional context**

brodergloosy, 2025-04-09:
**Additional context**[/-]
[+]Thanks a bunch![/+][-]

jneilliii, 2025-04-09:
I have a companion plugin that will store each bed level run. 

https://github.com/jneilliii/OctoPrint-BLVLogger

closed 2025-05-01 by github-actions[bot]